asukiaaa/MPU9250_asukiaaa #1: get gyro data

peiyzhou, 2018-02-01:
thanks for your sharing. Is it possible to add support for reading gyroscope data?

asukiaaa, 2018-02-03:
I add functions about gyrometer.
https://github.com/asukiaaa/MPU9250_asukiaaa/commit/e45174c4f45364a8bff32fe2e77a541c14dd95ef

You can use them after version 1.2.4.
How about them?

exzibith, 2018-02-05:
thanks, for sharing new sharing New version 1.2.4 library, but didn't found value of gyroscope data. value still showing 0,0,0. but on accel and mag data are normal showing value. 
i use wemos lolin32 board for pinout  I2C (21 &22).

asukiaaa, 2018-02-05:
I think that you can read value with using that board because I test my code with using [an ESP32 board](https://trialstacks.base.ec/items/6243312) and [a MPU9250 module](https://www.aliexpress.com/item/SPI-IIC-MPU9250-MPU-9250-MPU-9250-9-Axis-Attitude-Gyro-Accelerator-Magnetometer-Sensor-Module-MPU9250/32216818498.html?ws_ab_test=searchweb0_0,searchweb201602_2_10065_10068_10344_10342_10343_10340_10341_10084_10083_10618_10307_5722315_10313_10059_10534_100031_10103_441_10624_442_10623_10622_10621_10620_10142,searchweb201603_36,ppcSwitch_4&algo_expid=81af912e-51b3-4559-8ccf-1447d8881316-0&algo_pvid=81af912e-51b3-4559-8ccf-1447d8881316&transAbTest=ae803_4&priceBeautifyAB=0).

Do you call `beginGyro()` on setup?

If you have already done that, how about to add delay around begin methods like this.
```c
mySensor.setWire(&Wire);
delay(40);
mySensor.beginAccel();
delay(40);
mySensor.beginGyro();
delay(40);
mySensor.beginMag();
delay(40);
```

peiyzhou, 2018-02-05:
Thanks for the update! 
I have the same problem, the gyro still returns 0 after adding the delay for 40ms. I am using an Arduino Due and MPU 9250...

asukiaaa, 2018-02-05:
Oh, I missed break to set value to calculate gyro meter.
https://github.com/asukiaaa/MPU9250_asukiaaa/commit/5d90dbd67f2552806bdb2e76359d5f66a3771e2b

I cannot find this because I tested program before refactoring and releasing.
I found the problem with using released code, sorry.

You may use gyro value from version 1.2.5.

peiyzhou, 2018-02-06:
thanks, it works now!

closed 2018-02-06 by peiyzhou