asukiaaa/MPU9250_asukiaaa #2: set Offset formula

exzibith, 2018-02-05:
If you get values of sensor like this..
Name | Max | Min
-- | -- | --
magX | 70 | -20
maxY | 110 | 10

I suggest to set offset like this.
void setup() {
  mySensor.magXOffset = -30;
  mySensor.magYOffset = -60;
}
Then you can get like this.
Name | Max | Min
-- | -- | --
magX | 50 | -50
maxY | 50 | -50

======================================================
are this is code for calibrate range to determine Origin position area?

can you explain this code with other example?

sorry my bad english. 

thanks a lot.

japalmer, 2018-02-05:
I believe he's suggesting that you calculate a bias based on the mid point between the highest and lowest values.  In other words, the average of the MAX and MIN values for each axis.  Here's a great posting on the topic that includes code as an example: [https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration](url)

Best,
Jim

asukiaaa, 2018-02-05:
This program need offset value to average value become 0.
Graphs of [link](https://github.com/asukiaaa/MPU9250_asukiaaa/issues/url) that japalmer shared shows that.

I explain with other words because [value on my readme was wrong](https://github.com/asukiaaa/MPU9250_asukiaaa/commit/e9888e658e45ff73a7f385c1afa5dbb2b21a7715).

=============================

To get horizontal direction, I expect magnetometer values to satisfy the following requirement.

Requirement 1
```
applied_offset_x_max + applied_offset_x_min == 0 &&
applied_offset_y_max + applied_offset_y_min == 0
```

Applied offset values are calculated like this.

```
applied_offset_x_max = raw_x_max + x_offset;
applied_offset_x_min = raw_x_min + x_offset;
applied_offset_y_max = raw_y_max + y_offset;
applied_offset_y_min = raw_y_min + y_offset;
```

So we can think like this. (y_offset can get similar way.)

```
applied_offset_x_max + applied_offset_x_min == 0
-> (raw_x_max + x_offset) + (raw_x_min + x_offset) == 0
-> 2 * x_offset = - raw_x_max - raw_x_min
-> x_offset == - (raw_x_max + raw_x_min) / 2
```

Then you can get this requirement.

Requirement 2
```
x_offset == - (raw_x_max + raw_x_min) / 2 &&
y_offset == - (raw_y_max + raw_y_min) / 2
```

If you want to satisfy [Requirement 1], you need to adjust offset values with using [Requirement 2].

For example, if raw values become like this.
(You can get your sensor raw values on serial monitor by rotating the sensor.)
```
                [20, 110]
                   * ^^^ raw_y_max
    [-5, 103]*           *[55, 103]

 [-23, 75]*                *[63, 75]
            Rotate you sensor
[-30, 60]*         #         *[70, 60]
 ^^^ raw_x_min                 ^^ raw_x_max
 [-23, 35]*                *[63, 35]

     [-5, 17]*           *[55, 17]
                   *
                 [20, 10]
                      ^^ raw_y_min
```

In this case, you can calculate offset values with using [Requirement 2].

```
x_offset = - (70 + (-30)) / 2; // -20
y_offset = - (110 + 10)   / 2: // -60
```

Then you can set offset values like this.

```
mySensor.magXOffset = -20;
mySensor.magYOffset = -60;
```

After that you may get your sensor values like this.

```
                 [0, 50]
                   * ^^ raw_y_max
    [-25, 43]*           *[25, 43]

 [-43, 25]*                *[43, 25]
            Rotate you sensor
 [-50, 0]*         #         *[50, 0]
  ^^^ raw_x_min                ^^ raw_x_max
[-43, -25]*                *[43, -25]

   [-25, -43]*           *[25, -43]
                   *
                 [0, -50]
                     ^^^ raw_y_min
```

=============================

How about this explanation?

If it's not enough for you, could you explain what is not enough?

asukiaaa, 2018-02-13:
Close because no response for a week.

closed 2018-02-13 by asukiaaa