asukiaaa/MPU9250_asukiaaa #3: auto calculate offset

exzibith, 2018-02-21:
dear asukiaaa,

after i get update 1.30, now i can look value for gyrometer.
now i need help you, for calculate code for auto calculate offset even i start / power up esp32.
can i adapt your code for auto calculate offset value?
can you help me? do you have idea for auto calculate offset value even power up esp32?, like just standby the mpu9250 about 20second for calibrate the offset or rotate mpu9250 with different angle.

many thanx for your attention.
[image]

asukiaaa, 2018-02-21:
I cannot do this soon because I'm interested in other project.
If you add an example project and create pull request, I can review and give you some feedback for merging.

asukiaaa, 2018-03-05:
@exzibith 
How about this?
https://github.com/asukiaaa/MPU9250_asukiaaa/blob/master/examples/GetMagOffset/GetMagOffset.ino
You can use it after updating 1.3.1.

exzibith, 2018-03-08:
Wow, new update,... Thanks a lot for new example arduino code, bro. I will try this tonight. is this sketch can work properly without update library?

exzibith, 2018-03-10:
Oke, I have tried.. and it's work... But why the result value didn't get 0,0,0? I'm shake with O figure. Do I wrong procedure?

asukiaaa, 2018-03-11:
If I can help you, it's my pleasure.

> But why the result value didn't get 0,0,0

What you mean 0 ?
Offset values are expected to set to make each average values as 0.

> I'm shake with O figure. Do I wrong procedure?

Is it mean rotate 360 degree?
If yes, that is expected calibration.

By the way, what value do you want to get?
What is still unexpected value?

exzibith, 2018-03-12:
I'm rotate O figure with flat position board. Not rotate board to 360.
I want to get origin with 0 degree. If Mpu 9250 moved up to 10 degree. On serial output result get 10. If moved down, I get -10 degree.

I'm trying to create a project like the freedrum stick website (freedrum.rock), or hacked air drum with curie nano, i'm stuck 💡 idea, i'm using Wemos Lolin32 connected with Mpu 9250. Are the value magnetometer on this sketch can be used to my project? I want to get degree value. Like +- 0, 10, 50, 180 degree.

asukiaaa, 2018-03-13:
> I'm trying to create a project like the freedrum stick

Thank you, I understood what you want to do.
https://www.freedrum.rocks/

> I'm rotate O figure with flat position board. Not rotate board to 360.

If you don't rotate 360 degree, I don't understand about O figure.
(How do you rotate?)

> On serial output result get 10. If moved down, I get -10 degree.
> Are the value magnetometer on this sketch can be used to my project? I want to get degree value. Like +- 0, 10, 50, 180 degree.

I'm not sure that MPU9250 can satisfy your requirements, but I think that you need to custom `magHorizDirection()`.

`magHorizDirection()` uses x and y axis values (does not use z axis value).
It work so so for some situation that fixes z axis (likes on a car type robot).
However your case don't fix any axis so you need to consider which side faces to the ground.

It seems difficult.

Bharatmodugu, 2018-07-14:
Dear Asukia,
I want to control an RC car using the sensor readings. I am unable to understand what readings should I use to control the car. I want to get the readings in 0 to +/-360 degrees. Plz help in this regard.

Kendy545, 2020-08-02:
Hi @Bharatmodugu 

I'm just seeing that this discussion isn't yet close ! I've seen a very interresting  thing concerning your problem.

https://bodyinteraction.com/tag/mpu9250/