asukiaaa/MPU9250_asukiaaa #7: Get temperature of sensor

O60PMOT, 2019-06-14:
Hello, do you update to getting temperature of sensor?

O60PMOT, 2019-06-14:
i'm added in cpp file:
#define MPU9250_ADDR_TEMP_OUT_H   0x41
#define MPU9250_ADDR_TEMP_OUT_L   0x42

int16_t MPU9250::readTemp() {
  uint8_t temp;
  i2cRead(address, MPU9250_ADDR_TEMP_OUT_H, 2, &temp);
  return temp;
}

in h file:
  int16_t readTemp();

and i have 21.08 in result of:
Tmp = ((float) mySensor.readTemp()) / 333.87 + 21;
- but real temp is 26C

asukiaaa, 2019-06-14:
I also tried to use temperature before but it seems not accurate and heated bu itself.
There fore I don't add function to get temperature because it it is worthless.

O60PMOT, 2019-06-14:
i need to contol temp inside box, tell me please, where is my error in measuring degrees Celsius?

asukiaaa, 2019-06-14:
I don't know correct process to get temperature but it has ±4 difference.(3.1Gyroscope Specifications in [datasheet](http://www.invensense.com/wp-content/uploads/2015/02/PS-MPU-9250A-01-v1.1.pdf)) so your code might be correct.
I suggest you to use other module to measure temperature.

asukiaaa, 2019-06-14:
Sorry the `±4 difference` seems not mean the accuracy of temperature sensor.
I cannot find accuracy about that.

The following function is mentioned at `4.23Registers 65 and 66 –Temperature Measuremen` on [register map](http://www.invensense.com/wp-content/uploads/2017/11/RM-MPU-9250A-00-v1.6.pdf).
```
TEMP_degC = ((TEMP_OUT –RoomTemp_Offset)/Temp_Sensitivity) + 21degC
```

You may need some offset.

asukiaaa, 2019-06-14:
Your code may breaking memory.
You need to define length 2 uint8_t and merge them like this.
```c
int16_t MPU9250::readTemp() {
uint8_t temp[2];
  i2cRead(address, MPU9250_ADDR_TEMP_OUT_H, 2, &temp);
  return (((uint16_t)) temp[1] << 8 | temp[0]);
}
```

O60PMOT, 2019-06-14:
thank you! its works!

asukiaaa, 2019-06-15:
My pleasure.

closed 2019-06-26 by asukiaaa