asukiaaa/MPU9250_asukiaaa #8: M5StackのMAGデータ

nopnop2002, 2019-06-26:
こんにちは
M5Stackから以下のコードでMAGの値だけを取り出してみました。

	#include <MPU9250_asukiaaa.h>

	#ifdef _ESP32_HAL_I2C_H_
	#define SDA_PIN 21
	#define SCL_PIN 22
	#endif

	#define CALIB_SEC 20

	MPU9250 mySensor;

	uint8_t sensorId;
	float mDirection, mX, mY, mZ;

	void setup() {
	  while(!Serial);
	  Serial.begin(115200);
	  Serial.println("started");

	#ifdef _ESP32_HAL_I2C_H_ // For ESP32
	  Wire.begin(SDA_PIN, SCL_PIN); // SDA, SCL
	#else
	  Wire.begin();
	#endif

	  mySensor.setWire(&Wire);
	  mySensor.beginMag();
	  sensorId = mySensor.readId();

	}

	void loop() {
	 
	  mySensor.magUpdate();
	  mX = mySensor.magX();
	  mY = mySensor.magY();
	  mZ = mySensor.magZ();
	  Serial.print("at " + String(millis()) + "ms,");
	  Serial.print("sensorId: " + String(sensorId) + ",");
	  Serial.print("magX: " + String(mX) + ",");
	  Serial.print("maxY: " + String(mY) + ",");
	  Serial.println("magZ: " + String(mZ));
	  delay(500);
	}

結果は以下の様になりましたが、X-Yでグラフを書いても円になりません。
何か間違いがあるでしょうか？？

	at 108ms,sensorId: 113,magX: 1720.39,maxY: -1501.17,magZ: -445.59
	at 609ms,sensorId: 113,magX: 1722.77,maxY: -1502.36,magZ: -444.45
	at 1111ms,sensorId: 113,magX: 1718.01,maxY: -1501.17,magZ: -445.59
	at 1613ms,sensorId: 113,magX: 1720.39,maxY: -1501.17,magZ: -444.45
	at 2115ms,sensorId: 113,magX: 1720.39,maxY: -1501.17,magZ: -443.30
	at 2617ms,sensorId: 113,magX: 1722.77,maxY: -1502.36,magZ: -443.30
	at 3119ms,sensorId: 113,magX: 1722.77,maxY: -1503.55,magZ: -447.89
	at 3621ms,sensorId: 113,magX: 1727.54,maxY: -1519.04,magZ: -452.48
	at 4123ms,sensorId: 113,magX: 1729.92,maxY: -1541.68,magZ: -454.78
	at 4625ms,sensorId: 113,magX: 1728.73,maxY: -1553.59,magZ: -454.78
	at 5127ms,sensorId: 113,magX: 1728.73,maxY: -1573.85,magZ: -454.78
	at 5629ms,sensorId: 113,magX: 1720.39,maxY: -1586.95,magZ: -458.23
	at 6131ms,sensorId: 113,magX: 1716.82,maxY: -1577.42,magZ: -450.19
	at 6633ms,sensorId: 113,magX: 1714.43,maxY: -1582.19,magZ: -451.34
	at 7135ms,sensorId: 113,magX: 1710.86,maxY: -1594.10,magZ: -461.67
	at 7637ms,sensorId: 113,magX: 1691.80,maxY: -1602.44,magZ: -460.52
	at 8139ms,sensorId: 113,magX: 1684.65,maxY: -1600.06,magZ: -452.48
	at 8641ms,sensorId: 113,magX: 1682.27,maxY: -1597.68,magZ: -444.45
	at 9143ms,sensorId: 113,magX: 1678.69,maxY: -1598.87,magZ: -439.85
	at 9645ms,sensorId: 113,magX: 1675.12,maxY: -1603.63,magZ: -441.00
	at 10147ms,sensorId: 113,magX: 1666.78,maxY: -1596.48,magZ: -429.52
	at 10649ms,sensorId: 113,magX: 1667.97,maxY: -1595.29,magZ: -429.52
	at 11151ms,sensorId: 113,magX: 1671.54,maxY: -1596.48,magZ: -429.52
	at 11653ms,sensorId: 113,magX: 1675.12,maxY: -1608.40,magZ: -438.70
	at 12155ms,sensorId: 113,magX: 1665.59,maxY: -1619.12,magZ: -441.00
	at 12657ms,sensorId: 113,magX: 1662.01,maxY: -1627.46,magZ: -436.41
	at 13159ms,sensorId: 113,magX: 1653.67,maxY: -1626.27,magZ: -427.22
	at 13661ms,sensorId: 113,magX: 1653.67,maxY: -1629.84,magZ: -422.63
	at 14163ms,sensorId: 113,magX: 1648.91,maxY: -1628.65,magZ: -413.44
	at 14665ms,sensorId: 113,magX: 1642.95,maxY: -1632.23,magZ: -413.44
	at 15167ms,sensorId: 113,magX: 1642.95,maxY: -1631.04,magZ: -407.70
	at 15669ms,sensorId: 113,magX: 1633.42,maxY: -1626.27,magZ: -401.95
	at 16171ms,sensorId: 113,magX: 1627.46,maxY: -1609.59,magZ: -375.54
	at 16673ms,sensorId: 113,magX: 1615.55,maxY: -1611.97,magZ: -374.39
	at 17175ms,sensorId: 113,magX: 1611.97,maxY: -1609.59,magZ: -367.50
	at 17677ms,sensorId: 113,magX: 1595.29,maxY: -1591.72,magZ: -362.91
	at 18179ms,sensorId: 113,magX: 1597.68,maxY: -1588.14,magZ: -360.61
	at 18681ms,sensorId: 113,magX: 1596.48,maxY: -1586.95,magZ: -362.91
	at 19183ms,sensorId: 113,magX: 1606.02,maxY: -1603.63,magZ: -364.05
	at 19685ms,sensorId: 113,magX: 1614.36,maxY: -1582.19,magZ: -349.12
	at 20187ms,sensorId: 113,magX: 1619.12,maxY: -1565.51,magZ: -335.34
	at 20689ms,sensorId: 113,magX: 1615.55,maxY: -1520.23,magZ: -308.93
	at 21191ms,sensorId: 113,magX: 1629.84,maxY: -1536.91,magZ: -328.45
	at 21693ms,sensorId: 113,magX: 1646.52,maxY: -1528.57,magZ: -326.16
	at 22195ms,sensorId: 113,magX: 1658.44,maxY: -1525.00,magZ: -331.90
	at 22697ms,sensorId: 113,magX: 1673.93,maxY: -1522.62,magZ: -337.64
	at 23199ms,sensorId: 113,magX: 1682.27,maxY: -1527.38,magZ: -344.53
	at 23701ms,sensorId: 113,magX: 1702.52,maxY: -1535.72,magZ: -351.42
	at 24203ms,sensorId: 113,magX: 1713.24,maxY: -1547.64,magZ: -357.16
	at 24705ms,sensorId: 113,magX: 1721.58,maxY: -1554.79,magZ: -357.16
	at 25207ms,sensorId: 113,magX: 1727.54,maxY: -1561.93,magZ: -361.76
	at 25709ms,sensorId: 113,magX: 1733.50,maxY: -1569.08,magZ: -361.76
	at 26211ms,sensorId: 113,magX: 1735.88,maxY: -1577.42,magZ: -364.05
	at 26713ms,sensorId: 113,magX: 1738.26,maxY: -1589.34,magZ: -369.80
	at 27215ms,sensorId: 113,magX: 1735.88,maxY: -1595.29,magZ: -369.80
	at 27717ms,sensorId: 113,magX: 1735.88,maxY: -1596.48,magZ: -373.24
	at 28219ms,sensorId: 113,magX: 1734.69,maxY: -1598.87,magZ: -372.09
	at 28721ms,sensorId: 113,magX: 1732.30,maxY: -1596.48,magZ: -368.65
	at 29223ms,sensorId: 113,magX: 1725.16,maxY: -1577.42,magZ: -362.91
	at 29725ms,sensorId: 113,magX: 1722.77,maxY: -1575.04,magZ: -362.91
	at 30227ms,sensorId: 113,magX: 1723.96,maxY: -1572.66,magZ: -360.61
	at 30729ms,sensorId: 113,magX: 1723.96,maxY: -1571.46,magZ: -362.91
	at 31231ms,sensorId: 113,magX: 1723.96,maxY: -1573.85,magZ: -362.91

[image]

nopnop2002, 2019-06-26:
こんにちは
どうもデータのとり方を間違えていたようです。
M5Stackを垂直において、縦方向を軸として1回転させてみました。
XとZでグラフを書いたところ、扁平ですがほぼ円になりました。

[image]

asukiaaa, 2019-06-26:
期待通りに動いたようで良かったです。

nopnop2002, 2019-06-26:
こんにちは
返信、ありがとうございます。
とりあえずこのissuesをクローズしてください。

closed 2019-06-26 by asukiaaa