asukiaaa/MPU9250_asukiaaa #9: My NodeMCU is rebooting

matias1379, 2019-06-30:
Hi i have tried one of your examples code (GetData) for reading the  MPU9250 and my nodemcu start to rebooting 

i am very new so i don´t know what is happening rigth now

the serial says :

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

asukiaaa, 2019-07-01:
Can you find MPU9250 (address 0x68) by i2c_scanner?
https://playground.arduino.cc/Main/I2cScanner/

matias1379, 2019-07-03:
When i run i2c scan program the serial says :

Scanning...
No I2C devices found

asukiaaa, 2019-07-03:
It means your MPU9250 seems be not connected to your NodeMCU correctly.
Which MPU9250 device do you use?
I want to know that the device needs external pull up registers or has the registers inside.

matias1379, 2019-07-03:
I have just tried another i2c code and it says that there is a device 0x68 on the GPIO13 and GPIO12

matias1379, 2019-07-03:
On the front it says MPU-92/65 and on the back 9250

asukiaaa, 2019-07-03:
Is it?

[image]

If yes, you don't need any additional resister.

matias1379, 2019-07-03:
yes it is

matias1379, 2019-07-03:
i wired the D6 (SDA nodemcu) to the SDA 
and D7  to the SCL

asukiaaa, 2019-07-03:
If you can find devices by i2c_scanner except for MPU9250, there is a possibility of breaking the MPU9250 module.
If you have another module, you should try is.

matias1379, 2019-07-03:
i can find the MPU9250 with the i2c scanner but the GetData is still not working

asukiaaa, 2019-07-03:
I update this library not to handle values when it failed reading.
Can you try 1.4.2?
This updating will not solve this problem but may show more useful information on serial log.

matias1379, 2019-07-03:
I  tried 1.4.2 but the serial output is 

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

asukiaaa, 2019-07-04:
Do you run [GetData](https://github.com/asukiaaa/MPU9250_asukiaaa/blob/master/examples/GetData/GetData.ino)?

matias1379, 2019-07-04:
Yes I did

asukiaaa, 2019-07-05:
Does this program work on NodeMCU?
I don't have ESP8266 module now, so I cannot try this on it.

```c
#include <MPU9250_asukiaaa.h>

MPU9250_asukiaaa mySensor;

void setup() {
  Serial.begin(115200);
  delay(10);
  Wire.begin();
  delay(10);
  mySensor.setWire(&Wire);
}

void loop() {
  uint8_t sensorId;
  if (mySensor.readId(&sensorId) == 0) {
    Serial.println("sensorId: " + String(sensorId));
  } else {
    Serial.println("Cannot read sensorId");
  }
  Serial.println("at " + String(millis()) + "ms");
  Serial.println(""); // Add an empty line
  delay(500);
}
```

asukiaaa, 2019-07-05:
The code was written for version 1.5.x.

matias1379, 2019-07-05:
Yes that program work

the serial output is : sensor id :115 and the ms

matias1379, 2019-07-06:
Ey i have tried this code and it works 

#include <MPU9250_asukiaaa.h>

MPU9250_asukiaaa mySensor;

float aX, aY, aZ, aSqrt, gX, gY, gZ, mDirection, mX, mY, mZ;

void setup() {
  Serial.begin(115200);
  delay(10);
  Wire.begin(D6 , D7);
  delay(10);
  mySensor.setWire(&Wire);

  mySensor.beginAccel();
  mySensor.beginGyro();
  mySensor.beginMag();
}

void loop() {
  uint8_t sensorId;
  if (mySensor.readId(&sensorId) == 0) {
    Serial.println("sensorId: " + String(sensorId));
  } else {
    Serial.println("Cannot read sensorId");
  }

  if (mySensor.accelUpdate() == 0) {
    aX = mySensor.accelX();
    aY = mySensor.accelY();
    aZ = mySensor.accelZ();
    aSqrt = mySensor.accelSqrt();
    Serial.println("accelX: " + String(aX));
    Serial.println("accelY: " + String(aY));
    Serial.println("accelZ: " + String(aZ));
    Serial.println("accelSqrt: " + String(aSqrt));
  } else {
    Serial.println("Cannod read accel values");
  }

  
  if (mySensor.gyroUpdate() == 0) {
    gX = mySensor.gyroX();
    gY = mySensor.gyroY();
    gZ = mySensor.gyroZ();
    Serial.println("gyroX: " + String(gX));
    Serial.println("gyroY: " + String(gY));
    Serial.println("gyroZ: " + String(gZ));
  } else {
    Serial.println("Cannot read gyro values");
  }

  
  if (mySensor.magUpdate() == 0) {
    mX = mySensor.magX();
    mY = mySensor.magY();
    mZ = mySensor.magZ();
    mDirection = mySensor.magHorizDirection();
    Serial.println("magX: " + String(mX));
    Serial.println("maxY: " + String(mY));
    Serial.println("magZ: " + String(mZ));
    Serial.println("horizontal direction: " + String(mDirection));
  } else {
    Serial.println("Cannot read mag values");
  }
  
  Serial.println("at " + String(millis()) + "ms");
  Serial.println(""); // Add an empty line
  delay(500);
}

I dont know which is the main difference between this and GetData example because I copypasted it but now it works 

Thank you very much there are not much information about the mpu 9250 and you were my only hope 

I am very very happy now you are the best man

matias1379, 2019-07-06:
the problem was at the beginning of the void setup

while(!Serial);
Serial.begin(115200);

because the serial was off and the while was a infinity loop and the nodeMCU ran out of memory

closed 2019-07-06 by matias1379

asukiaaa, 2019-07-06:
I'm glad to know that you succeeded in running MPU9250 on NodeMCU.
Position of `while(!Serial)` caused ploblem so I change the position.
https://github.com/asukiaaa/MPU9250_asukiaaa/commit/8d02e39dbc7bc1a26bf970067b5a1aae42da87b7
Thank you for the information, the library was brushed up.