asukiaaa/MPU9250_asukiaaa #10: strange gyro behaviour

Attigliuzzo, 2019-07-03:
I'm building my head track with the hatire plugin using your library on opentrack and Debian. Why doesn't the gyro keep latest values? i'll explaine better, when i move the mpu9250 values of gyro change but after few seconds they return back to the horizontal tipical values ("drift effect"). It should keep values until i move my 9250.
A gyro must save current datas until i move it, is there something wrong in my usage of the mpu9250?? Same happens with your script "Get Data".
Thanks

asukiaaa, 2019-07-07:
I think that gyro is accelerator of wheel so they return to 0 after move.
https://en.wikipedia.org/wiki/Gyroscope
If you want to get current pose of sensor, please use accelerator values(accelX, accelY accelZ).
Or if you want to use last value of gyro, please hold the history on memory of Arduino or PC.

I'm not clear what you want but they are my opinion.

Attigliuzzo, 2019-07-07:
Thanks for you opinions, i'm investigating to get a code to drive my MPU91250. My first goal is to convert dinamic values (acceleration, Angolar speeds) in degrees using the time and Eulero's equations. Not  simple for a newbye, i''ll update you with my progresses.
Thanks to sharing and updating your library, it works very well

asukiaaa, 2019-07-08:
My pleasure.
Good luck.

Attigliuzzo, 2019-07-13:
Not simple, but i keep going on!! 
I test a strange behaviour only with acceleration datas, if i use this values:

> mySensor.accelUpdate();
> aX = mySensor.accelX();
>  aY = mySensor.accelY();
>  aZ = mySensor.accelZ();

I get zero values while using, from your script, an if statement like this:

> if (mySensor.accelUpdate() == 0) {
>     aX = mySensor.accelX();
>     aY = mySensor.accelY();
>     aZ = mySensor.accelZ();
> else {}

I get values, why "==0"? i think something like "!=0". If there are data ("!=0") it save values, else "do nothing". While with gyro or magnetometer data:

>     mySensor.gyroUpdate();
>     gX = mySensor.gyroX();
>     gY = mySensor.gyroY();
>     gZ = mySensor.gyroZ();

i receve datas, same with magnetic datas. Thanks

asukiaaa, 2019-07-14:
I can get these values with using current GetData program.

```
sensorId: 115
accelX: 0.03
accelY: 0.29
accelZ: -0.96
accelSqrt: 1.00
gyroX: 0.43
gyroY: 2.50
gyroZ: -0.06
magX: 42.25
maxY: 2.43
magZ: 35.04
horizontal direction: 86.71
at 14820ms
```

> I get values, why "==0"? i think something like "!=0".

*Update() functions returns value of [endTransmission()](https://www.arduino.cc/en/Reference/WireEndTransmission) so it means 0 as success.

Do you connect other i2c device on SDA and SCL line?

By the way, you call arduino c like program as `script` but I think it is not script.
shell and python program can run without compiling so they can be call as script but c program needs compiling to run so I think that arduino c like program should be call as `program` instead of `script`.

Attigliuzzo, 2019-07-14:
Now i connect it in the A4 and A5 pin instead of two other pins called "SCL" and "SDA", arduino doesn't freeze anymore and i get full data. 
Sorry for the mess

asukiaaa, 2019-07-15:
> Now i connect it in the A4 and A5 pin instead of two other pins called "SCL" and "SDA", arduino doesn't freeze anymore and i get full data.
I'm glad to know that.

Attigliuzzo, 2019-07-16:
Finally, things seem to go in the right direction!! i get stable values ￼ 

[image]

asukiaaa, 2019-07-16:
I don't understand details but good job.
If you want, you can crate a pull request with the program as an example of this project.

Attigliuzzo, 2019-07-17:
Yes but not now it is in a alpha state and needs some work. This is my plan:

datas:(accel., gyro, magnet.)-->fusion -->(pitch, roll, yaw)(*)--> 
-->
1. drive servos for a drone, heli, etc;
2. headtrack for flight sim.

I'm here (*). it isn't simple but i hope to get it working correctly.

asukiaaa, 2019-07-17:
Thank you for sharing your plan.
Good luck.

Dmytro-Bez, 2020-06-01:
Hi. There are questions. How to calibrate data when turning on the gyroscope and accelerometer?

Attigliuzzo, 2020-06-01:
I built my headtrack 3DOF using another library [this](https://github.com/hideakitai). This one gives to you yaw, pitch, roll and if you search in deep in his code you will find functions to calibrate acc, giro and magnetometer.

Dmytro-Bez, 2020-06-01:
Okay, but the devil will break his leg there - it's not clear. I sat for half a day disassembling.
I need smart advice. Please advise ...)

Attigliuzzo, 2020-06-02:
It is not simple, i suggest to study his examples and use his code. All what you need is there