asukiaaa/MPU9250_asukiaaa #12: maximum accel and gyro values get negative

josephlarralde, 2019-08-27:
I noticed that in the `MPU9250_asukiaaa::accelGet` and `MPU9250_asukiaaa::gyroGet` methods, the `-` sign applied to the bitwise operations before assignment to the int16_t value gets the maximum value to be negative.  
I managed to fix it by moving the `-` sign into the return statement, i.e:
```
  int16_t v = ((int16_t) gyroBuf[highIndex]) << 8 | gyroBuf[lowIndex];
  return ((float) -v) * gyroRange / (float) 0x8000;
```
I'm using an ESP32 but I suspect this bug is not related to the architecture.
Can you reproduce (and eventually fix) it ?

asukiaaa, 2019-08-28:
Thank you for the suggestion.
I merged your PR.
You can use updated code from v1.5.5.
https://github.com/asukiaaa/MPU9250_asukiaaa/releases/tag/1.5.5

josephlarralde, 2019-08-28:
I didn't try with other architectures than ESP32, and I'm no compiler specialist but it may be also related to my toolchain (I'm using platformio plugin in sublime).  
I think it is a harmless change anyway ...  
Thanks for the merge !

josephlarralde, 2019-08-28:
I confirm it works nicely with esp8266 and esp32, compiled and uploaded with platformio sublime plugin.

closed 2019-08-28 by josephlarralde