asukiaaa/MPU9250_asukiaaa #16: Using 2 MPU9250 sensors

iivoilov, 2020-02-02:
I need to read 2 sensors in one application. I connected them to different ports 4/5 and 2/0 for SDA/SCL. On initialization I set 2 different TwoWire objects and use them for 2 MPU9250_asukiaaa objects. When I read values from each MPU9250_asukiaaa I get the same values from both sensors, the values which come from sensor which was initialized the last.

Here is my initializing procedure
```C++
void init_mpu_sensor(MPU9250_asukiaaa& mpu, TwoWire* wire, int sda_pin, int scl_pin) {
  wire->begin(sda_pin,scl_pin);
  mpu.setWire(wire);
  mpu.beginAccel();
  mpu.beginGyro();
  mpu.beginMag();
}
```
Please check what can be wrong.

asukiaaa, 2020-02-02:
This flow seems no problem.
```c
  mpu.setWire(wire);
  mpu.beginAccel();
  mpu.beginGyro();
  mpu.beginMag();
```

Can you share wall of code about MPU9250 and information about arduino module that you are using?

asukiaaa, 2020-02-02:
I'm not sure it has meaning but how about this code?
```c
void init_mpu_sensor(MPU9250_asukiaaa* mpu, TwoWire* wire, int sda_pin, int scl_pin) {
  wire->begin(sda_pin,scl_pin);
  mpu->setWire(wire);
  mpu->beginAccel();
  mpu->beginGyro();
  mpu->beginMag();
}
```

iivoilov, 2020-02-02:
Attached my project file. The idea is to dump via wifi tcp connection readings from 2 MPU9650 sensors. I use NodeMCU board which is ESP8266 module. I use it from Arduino environment 1.8.11.
It has own implementation of Wire library, but interface is copying the one from AVR library.
I looked through code of Wire.cpp in both hardware implementation and see that twi_init() is quite different between them. So I suspect now that function beginWireIfNull() is causing causing incompatibility between hardwares

[read_2_mpu9259_wifi.zip](https://github.com/asukiaaa/MPU9250_asukiaaa/files/4143976/read_2_mpu9259_wifi.zip)

iivoilov, 2020-02-02:
I have overcome this problem by setting ports of Wire before reading

```C++
void process_sensors() {
  auto t = millis();
  if (t >= next_sensors) {
    next_sensors = t + 500;
    if (fsm > 0) {
      Wire1.begin(SDA_PIN1,SCL_PIN1);
      show_mpu_sensor_data("Sensor 1",mySensor1);
      Wire2.begin(SDA_PIN2,SCL_PIN2);
      show_mpu_sensor_data("Sensor 2",mySensor2);
    }
  }
}
```

closed 2020-02-02 by iivoilov

asukiaaa, 2020-02-04:
I'm glad to know that you have overcome the problem.
(Sorry, I did not notice the message with zip.)