asukiaaa/MPU9250_asukiaaa #17: Error Compilling for any Arduino Board

markwatermanlost, 2020-03-04:
Hi

Probably a simple thing I'm doing wrong but I get the message above and more error details as follows:

C:\Users\Admin\AppData\Local\Temp\cc5dfG77.ltrans0.ltrans.o: In function `setup':

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:51: undefined reference to `MPU9250_asukiaaa::setWire(TwoWire*)'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:53: undefined reference to `MPU9250_asukiaaa::readId(unsigned char*)'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:61: undefined reference to `MPU9250_asukiaaa::beginMag(unsigned char)'

C:\Users\Admin\AppData\Local\Temp\cc5dfG77.ltrans0.ltrans.o: In function `setMagMinMaxAndSetOffset':

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:89: undefined reference to `MPU9250_asukiaaa::magUpdate()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:91: undefined reference to `MPU9250_asukiaaa::magX()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:93: undefined reference to `MPU9250_asukiaaa::magY()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:95: undefined reference to `MPU9250_asukiaaa::magZ()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:103: undefined reference to `MPU9250_asukiaaa::magUpdate()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:105: undefined reference to `MPU9250_asukiaaa::magX()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:107: undefined reference to `MPU9250_asukiaaa::magY()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:109: undefined reference to `MPU9250_asukiaaa::magZ()'

C:\Users\Admin\AppData\Local\Temp\cc5dfG77.ltrans0.ltrans.o: In function `loop':

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:143: undefined reference to `MPU9250_asukiaaa::magUpdate()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:145: undefined reference to `MPU9250_asukiaaa::magX()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:147: undefined reference to `MPU9250_asukiaaa::magY()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:149: undefined reference to `MPU9250_asukiaaa::magZ()'

C:\Users\Admin\Documents\Arduino\sketch_mar04c/sketch_mar04c.ino:151: undefined reference to `MPU9250_asukiaaa::magHorizDirection()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Uno.

Any help much appreciated!!

asukiaaa, 2020-03-04:
What IDE are you using?
Arduino IDE, PlatformIO, or other?

markwatermanlost, 2020-03-04:
Arduino IDE

 

Nano board (although I’ve tried with Mega and Uno and same error).

 

Thanks

 

Mark

markwatermanlost, 2020-03-04:
Ahh

 

Found the issue I think. I had included your library as a file extension to the location not installed it.

 

i.e. I had: #include “C:\Users\Admin\Desktop\MPU9250_asukiaaa-master\src\MPU9250_asukiaaa.h"

 

Sorry for disturbing you!

 

Mark

asukiaaa, 2020-03-04:
I'm glad to know that you overcome the problem.

By the way, you can install this library with using library manager.
It can open by selecting [Sketch -> Include library -> Manage libraries].

closed 2020-03-04 by asukiaaa

markwatermanlost, 2020-03-04:
Thanks

 

I know about installing libraries but in my quest to find some code to do the following:

 

1.	Auto calibrate a MPU 9250 interfacing with a nano
2.	Read angles that do not drift

 

I have spent the last 40+ hour downloading various libraries and examples and some seem to have the same name but are not the same. I therefore assumed calling the library in the way described would get rid of this problem.

 

Does your code help me with my quest??

 

Many thanks

 

Mark