asukiaaa/MPU9250_asukiaaa #19: ESP32 support

Dmytro-Bez, 2020-05-12:
There was a problem.
Does your program support esp32? If so, then there are problems with compilation.
Please help solve this problem.

asukiaaa, 2020-05-13:
Thank you for reporting.
I updated definition of constructor to avoid errors for ESP32.
https://github.com/asukiaaa/MPU9250_asukiaaa/commit/3465938db0347f436e1f8393b4e6d12f5dc5c9b9
Please use version 1.5.8.

I supported ESP32 but some rule of compilation may be changed.

Dmytro-Bez, 2020-05-13:
> 3465938

Ok it works ))).
But I'm interested in a couple of questions:
1. Why such strange sensor data?
sensorId: 112
accelX: -0.67
accelY: -0.09
accelZ: -8.30
accelSqrt: 8.33
gyroX: 8.97
gyroY: -11.23
gyroZ: -1.46
Cannot read mag values
at 359824ms
2. I in the file MPU9250_asukiaaa.h changed the register data from 16G to
ACC_FULL_SCALE_2_G). But no changes. What is the problem?

Dmytro-Bez, 2020-05-29:
Hi. I have a question about the magnetometer on the mpu9250 sensor.
I took your project - everything works well. But my project needs a compass
to define north and south. Could you give advice on how to do it?

ср, 13 трав. 2020 о 10:13 Дмитро Безкоровайний <dzimitori2012@gmail.com>
пише:

asukiaaa, 2020-06-02:
If you need advice, share what you tried and code.
It's not enough to understand what you want to know.

Dmytro-Bez, 2020-06-02:
Okay. I organized the transmission of data (packets) via BLE, but only the
accelerometer data, now I am interested in calibrating the accelerometer,
gyroscope and magnetometer, and I want to transmit their data in a packet
via BLE.
The code is attached to the letter.
Thanks for the answer.

Dmytro-Bez, 2020-06-02:
I'm sorry the wrong file ... Here's a new one.

вт, 2 черв. 2020 о 13:37 Дмитро Безкоровайний <dzimitori2012@gmail.com>
пише:

asukiaaa, 2020-06-07:
I cannot see the code on issue page on github.
Can you paste on the [issue page](https://github.com/asukiaaa/MPU9250_asukiaaa/issues/19)?

Dmytro-Bez, 2020-06-07:
I have already solved my problem, thank you for your care ...

asukiaaa, 2020-06-08:
I'm glad to know that.

closed 2020-06-08 by asukiaaa

michel4731, 2022-07-11:
Hi.
this code it doesn't work in my esp32 board.

https://github.com/asukiaaa/MPU9250_asukiaaa/blob/master/examples/GetMagOffset/GetMagOffset.ino

asukiaaa, 2022-07-11:
How?
Show log or error.

michel4731, 2022-07-11:
Cannot find device to read sensorId

asukiaaa, 2022-07-11:
There are possibility to missing wiring or not supplying power to device.
Run i2c scanner and make sure that you can find 0x68 device.
https://github.com/asukiaaa/I2CScanner/blob/master/I2CScanner.ino

michel4731, 2022-07-11:
it is my connections
[image]

asukiaaa, 2022-07-11:
>checked my connects for several time
How do you connect?
I cannot see number of pins from the photo.

michel4731, 2022-07-11:
[image]

asukiaaa, 2022-07-11:
It seems no problem.
If you have other sensor or esp32, you should try with them because there is possibility of trouble of hardware.

asukiaaa, 2022-07-11:
My pleasure.
Also try other wires.
They sometimes cut and not connected.

michel4731, 2022-07-17:
hi dear Asuki Kono
I change my MPU and it work
thank you very much.

asukiaaa, 2022-07-17:
I'm glad to know that 😀

michel4731, 2022-07-18:
do you have sensor fusion for gy91?

asukiaaa, 2022-07-18:
I'm not sure about gy91.
I buy the module from aliexpress.
https://www.aliexpress.com/w/wholesale-mpu-9250.html

michel4731, 2022-07-18:
No. my main is complete the data of 9250 for quadcopter navigate.
for example Kalman ,  ANN, etc.

asukiaaa, 2022-07-18:
So?

michel4731, 2022-07-18:
I went to data fusion with 9250 for use in quadcopter.
I wish make flight controller

asukiaaa, 2022-07-18:
Good luck