asukiaaa/MPU9250_asukiaaa #22: Magnetometer reading error

GusakIurii, 2021-03-05:
started
sensorId: 113
accelX: -0.07
accelY: 0.00
accelZ: -1.02
accelSqrt: 1.02
gyroX: 0.85
gyroY: -1.59
gyroZ: 0.67
Cannot read mag values
at 102ms

sensorId: 113
accelX: -0.08
accelY: -0.02
accelZ: -1.01
accelSqrt: 1.02
gyroX: 0.79
gyroY: -1.59
gyroZ: 0.67
Cannot read mag values
at 610ms

https://www.banggood.com/MPU-9250-GY-9250-9-Axis-Sensor-Module-I2C-SPI-Communication-Board-Accelerometer-p-1227241.html?gmcCountry=UA&currency=UAH&cur_warehouse=CN&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=haosen&utm_campaign=haosen-ssc-ua-en-all-0204&ad_id=496617779448&gclid=Cj0KCQiAyoeCBhCTARIsAOfpKxg7xDlOaV53g41RPiJEQ3X8GkJnL23dzC8rwhLvgak8Ynwv_c5SNKEaAoHYEALw_wcB - board MPU9250
https://www.electrodragon.com/product/atmega328p-arduino-plus-lora-sx1278-board-loraduino/ - MCU borad Loraduino R1.2

I2C connection (A4, A5)
Arduino IDE 1.8.4

asukiaaa, 2021-03-05:
I want you to check these.

- Are you sure that it is MPU9250 not 6500?<br />MPU6500 does not have magnetometer and that kind board sometimes has 6500.
- Do you call [beginMag](https://github.com/asukiaaa/MPU9250_asukiaaa/blob/71e31d2021f29457ee9bf20fddaa851dc620bad2/examples/GetData/GetData.ino#L23)?<br />It is needed to activate magnetometer.

GusakIurii, 2021-03-05:
- I am using MPU9250
- Yes, I am doing mySensor.beginMag ();
I took the sketch - GetData.ino, no changes.

asukiaaa, 2021-03-05:
The module sometimes putted MPU6500.
I have ever got a MPU6500 module even if the site mentions only MPU9250.
Can you see engrave about MP95 like this?

This chip is written `MP95`.
[image]

If yes, I have no idea.
Can you try other module?

GusakIurii, 2021-03-05:
I managed to photograph the chip.
[image]

GusakIurii, 2021-03-05:
Ili MP92 W82L C1.
17 / 5000
Результаты перевода
Or MP92 W82L C1.  

[image]

GusakIurii, 2021-03-05:
My sensor works with the library https://github.com/bolderflight/mpu9250-arduino

This library is the only one that works with my hardware! But I would also like to get the coordinates of the direction of the sensor not in the axes, but in the tilt angle, and the second from the Magnetometer as the direction where it is looking.
I thought it was implemented in this library, but alas ..

asukiaaa, 2021-03-05:
Thank you for the photos.
Then I'm not sure the reason why.

Nodraak, 2021-03-19:
Hi,

I think the mag error comes from an error when reading the high/low bytes, they are switched:

https://github.com/asukiaaa/MPU9250_asukiaaa/blob/b983f4f47694ec7fdcd24ae29fc293fa175c6f46/src/MPU9250_asukiaaa.cpp#L151

vs

https://github.com/asukiaaa/MPU9250_asukiaaa/blob/b983f4f47694ec7fdcd24ae29fc293fa175c6f46/src/MPU9250_asukiaaa.cpp#L221

racs, 2021-05-18:
Same here can't read mag values, same board, one strange thing is that whoamI returns 0x70 and according to datasheet it's a mpu6050, I'm not sure if there is a wat to certify if it's a 9250 or 6050

sam17012k1, 2021-06-04:
> My sensor works with the library https://github.com/bolderflight/mpu9250-arduino
>
> This library is the only one that works with my hardware! But I would also like to get the coordinates of the [...]

Did you use it with esp32 only?

SebastianHD, 2021-07-24:
I have the same issue on a GY-91 board (contains the MPU9250). Using bolderflight, I read the magnetic sensor data just fine, but with this library, I cannot access it. Reading through the code, it looks like bolderflight uses master mode and reads the magnetic data through the MPU while this library tries to do the bypass and read the AK8963 directly. Unfortunately, I don't know enough right now to understand why that doesn't work.

narendar-yaga, 2022-04-13:
I have the same issue on the esp32 board. surly my sensor is mpu9250.
getting the response code is:117, I am reading accelerometer and gyro values, but mag values are not getting,
cannot read mag values message on the screen......

izsoandras, 2022-04-28:
Hi,
I tried with an ESP32 and Adafruit itsy bitsy 32u4, and this lib doesn't read magnetometer, I check and MP92 is written on my module. Bolder Flight lib doesn't work at all, it cannot communicate with the IMU. Did anyone figured out a workaround?

asukiaaa, 2022-04-29:
I checked and succeeded in reading values with using ESP32-WROOM-32D and [platform-espressif32.v4.1.0](https://github.com/platformio/platform-espressif32/releases/tag/v4.1.0) (it includes [arduino-esp32 v2.0.1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.1)).

Can you report with the following information?
- Environment with you use for building program.
- Hardware which you use.
- Result of [i2c-scanner](https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/Wire/examples/i2c_scanner/i2c_scanner.ino). Default device address of MPU9250 is 0x68.

In my case.
- platform-io 5.2.5 and platform-espressif32 4.1.0
- ESP32-WROOM-32D
- Succeeded in finding 0x68 device by i2c-scanner

marcofariasmx, 2022-05-30:
@izsoandras where you able to find a solution? I'm having the same issue and also a very similar serial number on the sensor board

izsoandras, 2022-06-14:
@marcofariasmx Unfortunately no. After I couldn't reach the magnetometer directly I gave up, because the exams kicked in. Maybe when they are finished I'll try it again.

asukiaaa, 2022-06-18:
I tried today and it's no problem on my environment.

My env
- PlatformIO 6.0.2
- platform-espressif 4.4.0 (It includes arduino-esp32 2.0.2)
- ESP-WROOM-32 (IC version 0)

>  Build: arduino IDE 1.8.19 with esp32 by Espressif Systems Version 1.0.6 library

I suggest you to use newer [arduino-esp32](https://github.com/espressif/arduino-esp32). Latest version is 2.0.3.

> Device: ESP-WROOM-32

It seems no problem.
I checked with using ESP-WROOM-32 (not 32D) which version of IC is 0.

>  I2C founds the 0x68 address. If I understand right, if I write 00000010 in the 0x37 register, I should be able to reach the magnetometer directly, which is not successful (on another board, with the same chip, according to the engraving, after writing this a 0x0C address shows up on the scan)

No, before that you need to write value to 0x6B register likes https://github.com/asukiaaa/MPU9250_asukiaaa/blob/b983f4f47694ec7fdcd24ae29fc293fa175c6f46/src/MPU9250_asukiaaa.cpp#L116-L122.

Overff, 2023-02-25:
have the same problem (can't read mag data)
sensorId: 112
accelX: 0.23
accelY: 0.03
accelZ: 1.03
accelSqrt: 1.06
gyroX: -8.06
gyroY: 1.95
gyroZ: 0.55
**Cannot read mag values 2**
at 3081ms

asukiaaa, 2023-02-25:
@Overff 
What IC (Arduino) do you use?

Overff, 2023-02-25:
> @Overff What IC (Arduino) do you use?

Actually i use Teensy3.6 (like arduino)
port A4, A5
[image]

asukiaaa, 2023-02-25:
My teensy3.6 could read mag value.
To read the mag values, calling beginMag is required.
Do you try restarting teensy 3.6 with connecting to MPU9250?
[image]

I add function to call beginMag again when failing to get mag values.
https://github.com/asukiaaa/MPU9250_asukiaaa/commit/c6882aecf94d3fd07800103aa5be197686e707ee

Overff, 2023-02-25:
Thy for your support. I have realised that your MPU is differ than mine.. Probably my has another register with mag data and so on. Could you provide some advice what should i need research to find out the problem?

asukiaaa, 2023-02-25:
Can you see `MP92` on your sensor IC as I mentioned on comment Mar 5, 2021?
https://github.com/asukiaaa/MPU9250_asukiaaa/issues/22#issuecomment-791453750
Sometimes board has MPU6060 which does not have magnetometer.

Overff, 2023-02-25:
> 

Checked it. Ii says: mp92/U601A1/2148
o_O

asukiaaa, 2023-02-25:
Thank you for the information.
It seems mpu9250.
I'm not sure the reason of this problem.

Overff, 2023-02-25:
thank you any way!

Wollo89, 2023-03-21:
I have the same sensor board as Overff "GY91 MPU9250+BMP280" and I also have the "MP92/U601A1/2148" installed on this board. Acceleration values and gyro values can be read out, but no values come from the magnetometer.

crackplatoonbd, 2023-04-05:
had anyone got any solution to read values come from the magnetometer ? written "mp92" on the sensor ic . but when i test with the program it shows 0x70 and MPU6500 .

Overff, 2023-05-20:
Hi there!
fyi,
Finally I coped with the problem. The problem is - fake board. My mpu-9250/6500 was a fake (. So, I've changed the board and problem dissipated. 

I2CScanner result (normal board):
--- Scan started ---
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
--- Scan finished ---

I2CScanner result (fake board):
--- Scan started ---
I2C device found at address 0x68  !
--- Scan finished ---

Actually they look the same.
[image]

brightproject, 2023-07-12:
> Hi there! fyi, Finally I coped with the problem. The problem is - fake board. My mpu-9250/6500 was a fake (. So, I've changed the board and problem dissipated.
>
> I2CScanner result (normal board): --- [...]

I don’t understand how the I2C scanner showed you the address of the magnetometer sensor 0x0C if this is an internal address.
The external address is 0x68, and the magnetometer sensor cannot be directly "reached out".
You are clearly lying.

antdavisonNZ, 2023-10-04:
https://invensense.tdk.com/wp-content/uploads/2015/02/PS-MPU-9250A-01-v1.1.pdf
page 24: Pass-Through mode is also used to access the AK8963 magnetometer directly from the host. In this
configuration the slave address for the AK8963 is 0X0C or 12 decimal.

antdavisonNZ, 2023-10-04:
I also purchased 5 gy-91s and they were also marked as "MP92/U601A1/2148"
All 5 were fake with a WHOAMI identifaction as 0x70, ie an mpu6500, 4 are pictured below, the fifth was already soldered to its carrier PCB and didnt scan well. The community needs to get together to identify and warn is members as to which marked 9250/5s are fake, and what behaviour to look for

[image]
[image]
[image]

AliExpress Item : https://www.aliexpress.com/item/32654511799.html

AliExpress Vendor : https://www.aliexpress.com/store/403088    
Far East Electronics
​Contact Person
Mr. Mars Wu
Telephone
86-755-82814528
Fax Number
86-755-82814200
Address
35/F World Trade Center, Causeway Bay,  Hong Kong

antdavisonNZ, 2023-10-12:
I re-purchased another 5 GY-91s which were genuine mp9250s (WHOAMI 0x71) with correctly working barometers, the previous purchase detailed above only 2 of the 5 BMP280s functioned correctly as well as having fake IMUs (WHOAMI 0x70)/ 
The genuine ones are at  
https://www.aliexpress.com/item/32591987418.html
[image]